linux-can/can-utils #569: slcan: how to set bitrate of canfd

zt9527, 2024-12-17:
When I used the slcand command to mount a Canable 2.0 device, I couldn't find a suitable command to set the bitrate of canfd.
When I use the cangaroo software, I found that it can set the bitrate of canfd . After checking its source code, it was found that there is a configuration instruction called "Y2/Y5", which is used to configure the bitrate of canfd .

[image]

`    // Set configured BRS rate
    if(_config.supports_canfd)
    {
        switch(_settings.fdBitrate())
        {
            case 2000000:
                _serport->write("Y2\r", 3);
                _serport->flush();
                break;
            case 5000000:
                _serport->write("Y5\r", 3);
                _serport->flush();
                break;
        }
    }`

marckleinebudde, 2024-12-17:
First things first: If you want to use the Canable 2.0 device with Linux I suggest to flash a candlelight/gs_usb compatible firmware. On the other hand I don't know if their candlelight firmware support CAN-FD.

If you want to add CAN-FD support to slcan, you have to figure out the available CAN-FD bitrates and corresponding commands. And in order to send CAN-FD messages the Linux kernel driver has to be extended, too. As far as I know, there is no proper standard for CAN-FD in the slcan protocol.

Have a look at this discussion: https://github.com/hardbyte/python-can/issues/1305, maybe this helps.

marckleinebudde, 2025-01-10:
Feel free to re-open the issue, if you want to extend the slcan in can-utils.

closed 2025-01-10 by marckleinebudde