linux-can/can-utils #570: pi5 mcp2518fd not working latest kernel

wrclark, 2024-12-18:
The hardware, a dual MCP2518FD hat works on Raspberry Pi 4 with the latest kernel, and strangely Raspberry Pi 5 with a 32-bit Raspberry Pi OS image, possibly for a Pi 4.

current not working setup:
```
[   27.385239] mcp251xfd spi0.1 can0: Controller failed to enter mode CAN 2.0 Mode (6) and stays in Configuration Mode (4) (con=0x068b0760, osc=0x00000468).
[   27.405210] mcp251xfd spi0.0 can1: Controller failed to enter mode CAN 2.0 Mode (6) and stays in Configuration Mode (4) (con=0x068b0760, osc=0x00000468).
```

```$ uname -r
6.6.66-v8-16k+
```
This took a Raspberry Pi OS 64-bit image from the Pi Imager, when that didn't work I did a `rpi-update` to the latest kernel which does not work either.

```
robin@raspberrypi:~ $ ./start_can.sh 
RTNETLINK answers: Connection timed out
RTNETLINK answers: Connection timed out
write: Network is down
read: Network is down
robin@raspberrypi:~ $ cat start_can.sh
#!/usr/bin/sh

sudo ip link set can0 type can bitrate 500000
sudo ip link set can1 type can bitrate 500000

sudo ip link set can0 up
sudo ip link set can1 up

cangen can0 & 
candump can1
robin@raspberrypi:~ $ 
```

mishka, 2024-12-18:
I'm experiencing the same problem. Any help would be appreciated.

marckleinebudde, 2024-12-18:
> The hardware, a dual MCP2518FD hat works on Raspberry Pi 4 with the latest kernel, and strangely Raspberry Pi 5 with a 32-bit Raspberry Pi OS image, possibly for a Pi 4.

Sorry I'm struggling to follow. Can you make a list of what's working and what's broken. The list should include: Pi version, Kernel Version, 32/64 bit.

> ```
> [   27.385239] mcp251xfd spi0.1 can0: Controller failed to enter mode CAN 2.0 Mode (6) and stays in Configuration Mode (4) (con=0x068b0760, osc=0x00000468).
> [   27.405210] mcp251xfd spi0.0 can1: [...]
Do you get the same error message with all broken kernel versions?

Please post the output of `dmesg | grep spi`

Please post the output of:
```
ip --details -s -s -s link show can0
```

What's you `config.txt` snippet to activate the `mcp251xfd`?

wrclark, 2024-12-18:
I will post Pi 5 debug later tomorrow, but for now here is what works if that helps at all:

OS: Debian GNU/Linux 12 (bookworm) aarch64
Host: Raspberry Pi 4 Model B Rev 1.5
Kernel: 6.6.64-v8+

`/boot/firmware/config.txt`
```
dtparam=spi=on
dtoverlay=mcp251xfd,spi0-0,interrupt=25,oscillator=20000000
dtoverlay=mcp251xfd,spi0-1,interrupt=26,oscillator=20000000
```
(same on Pi 5)

```
$ dmesg | grep spi
[    6.868528] spi_master spi0: will run message pump with realtime priority
[    6.877502] mcp251xfd spi0.1 can0: MCP2518FD rev0.0 (-RX_INT -PLL -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD o:20.00MHz c:20.00MHz m:20.00MHz rs:8.50MHz es:8.33MHz rf:8.50MHz ef:8.33MHz) successfully initialized.
[    6.884036] mcp251xfd spi0.0 can1: MCP2518FD rev0.0 (-RX_INT -PLL -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD o:20.00MHz c:20.00MHz m:20.00MHz rs:8.50MHz es:8.33MHz rf:8.50MHz ef:8.33MHz) successfully initialized.
```

```
 $ ip --details -s -s -s link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 72 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0  allmulti 0 minmtu 0 maxmtu 0
    can <FD> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 0
          bitrate 500000 sample-point 0.875
          tq 50 prop-seg 17 phase-seg1 17 phase-seg2 5 sjw 2 brp 1
          mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp_inc 1
          dbitrate 1000000 dsample-point 0.750
          dtq 50 dprop-seg 7 dphase-seg1 7 dphase-seg2 5 dsjw 2 dbrp 1
          mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp_inc 1
          clock 20000000
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus spi parentdev spi0.1
    RX:  bytes packets errors dropped  missed   mcast
           251      45      0       0       0       0
    RX errors:  length    crc   frame    fifo overrun
                     0      0       0       0       0
    TX:  bytes packets errors dropped carrier collsns
           251      45      0       0       0       0
    TX errors: aborted   fifo  window heartbt transns
                     0      0       0       0       1
```

marckleinebudde, 2024-12-19:
Thanks, this looks quite normal. Are you sure about the `20 MHz` oscillator? It's a bit uncommon.

marckleinebudde, 2024-12-19:
Is your device connected to the CAN bus during `ip link set can0 up`?

wrclark, 2024-12-19:
> Thanks, this looks quite normal. Are you sure about the `20 MHz` oscillator? It's a bit uncommon.

Yes it all works on the Pi 4

> Is your device connected to the CAN bus during `ip link set can0 up`?

Yes

marckleinebudde, 2024-12-19:
Can you try to disconnect the device from the bus before `ip link set can0 up` on the Pi 5?

marckleinebudde, 2024-12-19:
>> Thanks, this looks quite normal. Are you sure about the 20 MHz oscillator? It's a bit uncommon.

> Yes it all works on the Pi 4

Have you tested it against a different CAN controller? I'm asking because you will not notice that problem if you have the same error on both sides of the CAN bus.

wrclark, 2024-12-19:
> Can you try to disconnect the device from the bus before `ip link set can0 up` on the Pi 5?

That seemed to fix it, it's all working now.

> > > Thanks, this looks quite normal. Are you sure about the 20 MHz oscillator? It's a bit uncommon.
> 
> > Yes it all works on the Pi 4
> 
> Have you tested it against a different CAN controller? I'm asking because you will not notice that problem if you have the same error on both sides of the CAN bus.

The HAT has two mcp2518fd (+ transceiver). I am connecting those two together.

What do you think caused this? I configured + up both can ip dev while they were connected to eachother and it worked without a problem

marckleinebudde, 2024-12-20:
Can you reproduce the problem on the Pi 5, if you connect the 2 mcp2518fd and bring up both interfaces without sending any data before both are up?

Please send me the output of `dmesg | grep spi`, even if it works.

If you can reliable reproduce the problem and you're familiar on how to compile a kernel for the Pi or want to learn it, try if the following patch helps. It basically increases to timeout to 100ms:

```
diff --git a/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c b/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c
index 3bc56517fe7a..cfe51dd4ab5e 100644
--- a/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c
+++ b/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c
@@ -253,6 +253,8 @@ __mcp251xfd_chip_set_mode(const struct mcp251xfd_priv *priv,
                                    MCP251XFD_FRAME_LEN_MAX_BITS * USEC_PER_SEC /
                                    bt->bitrate);
 
+        timeout_us = 100 * USEC_PER_MSEC;
+
         err = regmap_read_poll_timeout(priv->map_reg, MCP251XFD_REG_CON, con,
                                        !mcp251xfd_reg_invalid(con) &&
                                        FIELD_GET(MCP251XFD_REG_CON_OPMOD_MASK,
```

Regarding the oscillator, do you have the schematics of your Hat to check which oscillator is equipped on that board. Maybe you can post a pic of the board here in high resolution and we can try to figure out.

marckleinebudde, 2024-12-22:
> Everything seems to work... Perhaps when I call it in the script it does it too quickly and the bus gets held in some funny state? Not sure.

The mcp2518fd needs to see an idle bus when it changes from config- to CAN-2.0 or CAN-FD mode. Time time it takes obviously depends on the bit rate of the bus and the length of the CAN frame currently transmitted. While writing the driver I didn't pay attention to this detail and picked IIRC 1ms out of thin air, and it worked (for me). Later we made the timeout dependent on the bit rate https://github.com/torvalds/linux/commit/9efa1a5407e81265ea502cab83be4de503decc49, maybe the timeout is still too small.

> I know it's a 20 MHz xtal I laid it out in kicad myself. If it ever starts happening again ill try to compile it myself but I don't have a super fast computer so it will take a couple of hours I imagine, thanks for your help

Thanks for the explanation. 20 MHz is unusual, as the SPI bus speed is limited by the oscillator. If you configure the wrong oscillator the communication should work, if you have the same error on the receiving and sending side. But the timeout calculation mentioned above would be wrong.

> Also this board is a test/dev revision so final one will have a low ppm smd xtal prob 40 mhz. Currently got HC49 size 20 mhz that might drift too much for normal operation?

I'm more a software guy, but having a better clock is always better for CAN, especially CAN-FD operations.

Can we close this issue?

wrclark, 2024-12-22:
The init issue has not happened yet again, so I will put a low ppm 40 MHz on the next version. 

By the way, is it possible, via the pre packaged dts overlay file, to enable tx/rx intterrupts on the gpio/int lines on the mcp2518fd? I had a look and can not figure out how to

You can close this issue if you'd like, thanks !

closed 2024-12-23 by marckleinebudde