linux-can/can-utils #572: Make a new release

MarkAtBosch, 2025-01-22:
I would like to propose to make a new release. There is quite a number of commits on master since the last [release v2023.03](https://github.com/linux-can/can-utils/releases/tag/v2023.03).

If there was release in the next days / few weeks it would still be possible to update the [meta-openembedded can-utils recipe](https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-extended/socketcan) and get it into the next [Yocto Project / OpenEmbedded release in April 2025](https://wiki.yoctoproject.org/wiki/Releases).

hartkopp, 2025-01-24:
Hi Mark,

a good suggestion.

The current master repo also has the full CAN XL support and I sent a pull request for the last missing CAN XL piece (ASC file converstion tools) right now: https://github.com/linux-can/can-utils/pull/573

I don't expect any relevant problems showing up and I will test the new canbusload features too.

Creating a new release at the beginning of next week seems reasonable then.

hartkopp, 2025-01-28:
Done ;-)

https://github.com/linux-can/can-utils/releases/tag/v2025.01

Thanks Mark!

closed 2025-01-28 by hartkopp

marckleinebudde, 2025-01-28:
Thanks @hartkopp for the detailed summary!