linux-can/can-utils #574: ECC Error detected

pompushko, 2025-01-31:
I'm not sure, if that is hardware issue or nope. I hope that is not a hardware issue :D

In some cases, reboot helps, in some cases `ifconfig can0 down` and `ifconfig can0 up`
Also, unfortunately, thats happens only with mcp251xfd which is connected to spi1-0

```
# CAN-bus shield
dtoverlay=spi1-3cs
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi0-1,interrupt=13
dtoverlay=mcp251xfd,spi1-0,interrupt=24
```

```
filename:       /lib/modules/6.6.62+rpt-rpi-v7/kernel/net/can/can.ko.xz
alias:          net-pf-29
author:         Urs Thuermann <urs.thuermann@volkswagen.de>, Oliver Hartkopp <oliver.hartkopp@volkswagen.de>
license:        Dual BSD/GPL
description:    Controller Area Network PF_CAN core
srcversion:     9FCDE9DC80F2506E3093972
depends:
intree:         Y
name:           can
vermagic:       6.6.62+rpt-rpi-v7 SMP mod_unload modversions ARMv7 p2v8
parm:           stats_timer:enable timer for statistics (default:on) (int)
```

```
Linux device 6.6.62+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.62-1+rpt1 (2024-11-25) armv7l GNU/Linux
```

```
[  276.067936] mcp251xfd spi1.0 can-c1: Double ECC Error detected at address 0x0470.
[  287.749071] mcp251xfd spi1.0 can-c1: Single ECC Error detected at address 0x0470.
[  293.302682] mcp251xfd spi1.0 can-c1: Single ECC Error detected at address 0x047c.
[  299.193674] mcp251xfd spi1.0 can-c1: Single ECC Error detected at address 0x0534.
[  316.325433] mcp251xfd spi1.0 can-c1: Single ECC Error detected at address 0x0564.
```

marckleinebudde, 2025-02-02:
Can you send me a register dump of the mcp251xfd? Please configure the interface as usual, but it doesn't have to be in the faulty state. It's the file `/sys/kernel/debug/regmap/spi1.0-crc/registers`.

marckleinebudde, 2025-02-02:
Here the mcp251xfd device detects an ECC error in the on chip RAM and throws an interrupt. Do you see any corrupt or missing RX'ed CAN frames?

pompushko, 2025-02-03:
```
[884992.979422] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979465] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979495] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979499] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000).
[884992.979529] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0004 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979559] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0004 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979588] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0004 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979590] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0004 (length=4, data=00 00 00 00, CRC=0x0000).
[884992.979620] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0008 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979649] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0008 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979677] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0008 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979680] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0008 (length=4, data=00 00 00 00, CRC=0x0000).
[884992.979710] mcp251xfd spi1.0 can-c1: CRC read error at address 0x000c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979738] mcp251xfd spi1.0 can-c1: CRC read error at address 0x000c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979767] mcp251xfd spi1.0 can-c1: CRC read error at address 0x000c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979770] mcp251xfd spi1.0 can-c1: CRC read error at address 0x000c (length=4, data=00 00 00 00, CRC=0x0000).
[884992.979800] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0010 (length=4, data=80 00 00 00, CRC=0x0000) retrying.
[884992.979829] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0010 (length=4, data=80 00 00 00, CRC=0x0000) retrying.
[884992.979858] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0010 (length=4, data=80 00 00 00, CRC=0x0000) retrying.
[884992.979861] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0010 (length=4, data=80 00 00 00, CRC=0x0000).
[884992.979890] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0014 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979919] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0014 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979948] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0014 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.979951] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0014 (length=4, data=00 00 00 00, CRC=0x0000).
[884992.979980] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0018 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980009] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0018 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980037] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0018 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980040] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0018 (length=4, data=00 00 00 00, CRC=0x0000).
[884992.980070] mcp251xfd spi1.0 can-c1: CRC read error at address 0x001c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980099] mcp251xfd spi1.0 can-c1: CRC read error at address 0x001c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980127] mcp251xfd spi1.0 can-c1: CRC read error at address 0x001c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980130] mcp251xfd spi1.0 can-c1: CRC read error at address 0x001c (length=4, data=00 00 00 00, CRC=0x0000).
[884992.980180] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0020 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980210] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0020 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980240] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0020 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980243] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0020 (length=4, data=00 00 00 00, CRC=0x0000).
[884992.980273] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0024 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980301] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0024 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980330] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0024 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980333] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0024 (length=4, data=00 00 00 00, CRC=0x0000).
[884992.980364] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0028 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980393] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0028 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980422] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0028 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[884992.980424] mcp251xfd spi1.0 can-c1: CRC read error at address 0x0028 (length=4, data=00 00 00 00, CRC=0x0000).
[... cut ...]
02c: 00000000
030: 00000000
034: 00200000
038: 00000000
03c: 00000000
040: 00000400
044: 00000000
048: 00000000
04c: 00000000
050: 00600480
[... 661 lines cut ...]
bb8: aefd1bb0
bbc: 96bd4fa6
bc0: fa8cfcef
bc4: dfb771cb
bc8: 5c6aff55
bcc: 44a7d3f4
bd0: 751e7cc0
bd4: a861a784
bd8: ae72caed
bdc: 4d63c39b
be0: 4b83375e
be4: 04416dd8
be8: d9cb25a3
bec: ade974be
bf0: 45dcb93a
bf4: 56e99416
bf8: 3bbc363d
bfc: b1563058
e00: 00000460
e04: 03000003
e08: 00000000
e0c: 00000000
e10: 00000000
e14: 00000000
```