linux-can/can-utils #575: Response handling with ISO-TP and functional addressing (broadcasts) ?

Comer352L, 2025-02-08:
After reading lot's of stuff about the SocketCAN-API and ISOTP, I am searching for examples concerning functional addressing (broadcasts).
I know flag CAN_ISOTP_SF_BROADCAST can be used for that, but it's not clear to me how incoming messages can be separated and how to find out who has sent them.
Is that possible at all ?

Communication scenario:
-> functional request message is send out
<- existing devices on the bus respond, providing individual information about their capabilities etc.
-> physical requests to individual communication partners (different tx can ids)
<- single or periodic responses

hartkopp, 2025-02-08:
Hello @Comer352L ,
the ISOTP protocol in the Linux kernel generally implements the bidirectional segmentation protocol using two CAN IDs.

Functional addressing (broadcast) was formerly implemented with a CAN_RAW socket.
The flag CAN_ISOTP_SF_BROADCAST is just a convenience functionality to build correct ISOTP PDUs and to stay in the CAN_ISOTP socket universe.

Therefore you can send functional addressing requests with a socket opened with the CAN_ISOTP_SF_BROADCAST flag.
And to talk to the individual communication partners you need to open additional (multiple) CAN_ISOTP sockets with the specific RX/TX CAN IDs pairs. In C you can listen on multiple ISOTP sockets e.g. with the `select()` or `epoll()` syscall.

I'm not sure whether there are "functional request responses" or whether the ECUs directly answer on their assigned RX/TX CAN ID pair? Because you need to know on which CAN IDs these answers are sent.

Comer352L, 2025-02-09:
Thanks for your help.

> I'm not sure whether there are "functional request responses" or whether the ECUs directly answer on their assigned RX/TX CAN ID pair? Because you need to know on which CAN IDs these answers are sent.

The responses are physically addressed, but the CAN IDs are not known at communication session start.
Well, there's of course a range of possible addresses/IDs, but it is too large to probe them all.

AFAIU, when setting the CAN_ISOTP_SF_BROADCAST flag, no Rx address is set/used.
So I would expect to receive all responses on this socket.
Or do I have to set up 1 socket for each possible response ID ??

hartkopp, 2025-02-09:
> Thanks for your help.
>
>
> The responses are physically addressed, but the CAN IDs are not known at communication session start. Well, there's of course a range of possible addresses/IDs, but it is too [...]

Sure?  E.g. in a vehicle the RX/TX addresses (physical addressing) are defined for each ECU. So once you know which ECU is available in the vehicle you should also know the physical addresses.

> AFAIU, when setting the CAN_ISOTP_SF_BROADCAST flag, no Rx address is set/used. 

Correct!

> So I would expect to receive all responses on this socket.

No. When there is no RX address defined you get nothing on `read()`.
Broadcast means broadcast - as you know it from a radio station.

> Or do I have to set up 1 socket for each possible response ID ??

Yes. You need to create one socket for each physical address pair.
Of course you can open hundreds of sockets which just hang around until they  receive ISOTP traffic ;-)

hartkopp, 2025-02-11:
> Now, what if the ECU sends multiple/periodic responses ? Any chance to get them separated without having to apply knowledge about the next upper protocol layer ?

An ECU may have different 'ECU entities' inside (e.g. a Navigation and a Radio inside a single box). This would lead to two independent physical addresses and communication relations.

But usually multiple periodic responses are defined inside UDS while talking to two different UDS services inside a single ISOTP connection.

> With datagram sockets, can I assume that only a single EPOLLIN event is triggered per received datagram/message and that it is triggred AFTER message reception is complete (as long as EPOLLET isn't set) ?

Yes. You always get a complete ISOTP PDU.

> Then reading the bytes immediately after the event has been received should work. At least as long nothing causes too large delays, but I think the kernel can't be that busy. ;-)

The kernel is not the limiting factor. It is either the ECU response time or the data transmission time on the CAN bus.

Comer352L, 2025-02-13:
> An ECU may have different 'ECU entities' inside (e.g. a Navigation and a Radio inside a single box). This would lead to two independent physical addresses and communication relations.
>
> But usually [...]

For simplicity, let's take OBD as an example, with a single ECU:
If the ECU doesn't have the requested data availbale within P2 timing, it may send one or multiple "response pending" responses followed by a final response.
Can I separate these messages with the ISO 15765 layer (and omit applying knowledge from the OBD application layer) ?

With SOCK_RAW / CAN_RAW, I'm reading fixed size structs (can_frame or canfd_frame), so it is easy to separate the frames.
But with SOCK_DGRAM / CAN_ISOTP I don't know how many bytes belong to a PDU.

hartkopp, 2025-02-14:
> > > With datagram sockets, can I assume that only a single EPOLLIN event is triggered per received datagram/message and that it is triggred AFTER message reception is complete (as long as EPOLLET isn't set) ?
> > 
> > 
> > Yes. You always get a complete ISOTP PDU.
> > > Then reading the bytes immediately after the event has been received should work. At least as long nothing causes too large delays, but I think the kernel can't be that busy. ;-)
> > 
> > 
> > The kernel is not the limiting factor. It is either the ECU response time or the data transmission time on the CAN bus.
> 
> Well, you actually answered my question with this, sorry. So it it safe to simply do poll - get event - read all bytes available = single message - poll ...

Yes.

> Bonus question: For detection of response timeouts (application layer), I also need to know when reception of the first frame has has started. Will the follwing work ?: poll for an edge-triggered EPOLLIN event - switch to non-edge-triggered polling (epoll_ctl(EPOLL_CTL_MOD, ...)) - poll for the message completion

Bonus answer: No. :-D

You get an event, when the PDU is completely received by the kernel (similar to UDP). And then you read the entire PDU in one chunk.

When the ECU does not answer in time you need an application layer timeout.
When the ECU started to send an answer PDU but stops sending data (for whatever reason) the ISO-TP protocol timeout (1000 ms)  will detect the missing data flow and the socket returns the ETIMEDOUT error.

Comer352L, 2025-02-16:
> Bonus answer: No. :-D
>
> You get an event, when the PDU is completely received by the kernel (similar to UDP). And then you read the entire PDU in one chunk.

Does that mean that EPOLLET (edge-triggered) flag will not work at all ?

> When the ECU does not answer in time you need an application layer timeout. 

Yes, but the timeout is usually measured until message reception start and doesn't include the message transfer time.
That's why e.g. the SAE-J2534 userspace API provides indication messages with a Rx status flag ISO15765_FIRST_FRAME / START_OF_MESSAGE for that, which are returned immediately after recpetion start.

But yes, you can usually clobber that by using a much larger timout value than specified. ;-)

> When the ECU started to send an answer PDU but stops sending data (for whatever reason) the ISO-TP protocol timeout (1000 ms) will detect the missing data flow and the socket returns the ETIMEDOUT error.

Ok, thanks for this extra info.

I think I will start with a simple OBD request-response program as soon as my CANable device has arrived (and outside temperatures have increased ;-))

Many thanks for help !

hartkopp, 2025-02-17:
> > You get an event, when the PDU is completely received by the kernel (similar to UDP). And then you read the entire PDU in one chunk.
> 
> Does that mean that EPOLLET (edge-triggered) flag will not work at all ?

I did not test that in my own. But the trigger would fire only when the PDU is completely received (or when a timeout occurred) anyway.

> Yes, but the timeout is usually measured until message reception start and doesn't include the message transfer time.
> That's why e.g. the SAE-J2534 userspace API provides indication messages with a [...]

Right. The idea was to follow standard IT patterns you know from Linux/Unix network programming. So this very embedded concept was intentionally omitted as we handle the ISOTP PDU analog to a UDP/IP PDU without looking at the segmentation into CAN frames on the bus.

The timestamp which is tagged to the ISOTP PDU is from the last data CAN frame which was received.

> But yes, you can usually clobber that by using a much larger timout value than specified. ;-)

In any case you would need an estimation about the time the entire PDU needs on the bus right?

So there is
- the ECU reaction time before it starts to reply
- the needed time for the PDU on the (loaded?) bus
- a potential 1000ms timeout when the ECU crashed

My assumption is, that you always need to get an impression about the needed time on the bus to specify a proper timeout.

If you would like to make timestamping measurements from the first frame to the last frame of the ISO-TP PDU, you can e.g. use the CAN_BCM protocol and filter for all incoming FF messages on the receiving ISOTP CAN ID. With that you can get the timestamp from the FF.

> Ok, thanks for this extra info.
>
> I think I will start with a simple OBD request-response program as soon as my CANable device has arrived (and outside temperatures have increased ;-))
>
> Many thanks [...]

Have fun!

closed 2025-02-19 by hartkopp