linux-can/can-utils #578: Incomplete public header set

jengelh, 2025-02-10:
Version: 2025.01

As per CMakeLists.txt, cmake installs some header files

```
  set(PUBLIC_HEADER_ISOBUSFS
    isobusfs/isobusfs_cmn.h
    isobusfs/isobusfs_cmn_cm.h
  )

  set_target_properties(isobusfs PROPERTIES
    PUBLIC_HEADER "${PUBLIC_HEADER_ISOBUSFS}"
  )
```

but these are useless in practice because they depend on other, uninstalled, files:

```
In file included from x.cpp:1:
/usr/include/isobusfs_cmn.h:13:10: fatal error: ../libj1939.h: No such file or directory
   13 | #include "../libj1939.h"
      |          ^~~~~~~~~~~~~~~
```

marckleinebudde, 2025-03-01:
That's not good. The isobus code is licensed under LGPL-2.1 and uses some code of the libj1939, which is GPL-2.0.

@olerem do you only form the libj1939 that you have written? Then move it out into a separate lib under LGPL-2.1....otherweise we might need to re-license the whole lib???

kurt-vd, 2025-03-04:
License the whole lib to GPL-2.0 or LGPL-2.1?
I think the code in the original libj1939.c wasn't so special that it deserved a shared library.
The Makefile confirms my memory. I see no library being built.
In the case without a shared library, is there a residual difference between GPL-2.0 and LGPL-2.1?

marckleinebudde, 2025-03-04:
Here's the history of ` libj1939.c` https://github.com/linux-can/can-utils/commits/master/libj1939.c . This code was written by you.

As always IANAL:

LGPL-2.1 allows linking from proprietary code. While GPL-2.0 doesn't.

> In the case without a shared library, is there a residual difference between GPL-2.0 and LGPL-2.1?

One can convert any LGPL-2.1 Code into a shared library and link from proprietary code against it, as long a you provide the sources, i.e. comply with the license.

jengelh, 2025-03-04:
A core idea of Library-type licenses is that the Combined Work can be recreated with a {user-modified Library part built from source}. This can also be accomplished with static libraries and static linking, it's just a different challenge (mostly on how to exercise tooling).

kurt-vd, 2025-03-07:
I'm (1) not against re-licensing
but I do not think that the (original) code of mine (2) deserves to be a shared library, with explicit API description, including complete semantic behavior.