linux-can/can-utils #580: J1939: MSG_DONTROUTE not set on segmented messages of local origin

xile273, 2025-02-18:
The Socket-Can stack sets the MSG_DONTROUTE flag when a received frame originates from the local host.

For simple, non-segmented, J1939 messages the same is true. When receiving such a message of local origin, the MSG_DONROUTE
flag is set. When receiving segmented messages originating from the local host however, the MSG_DONTROUTE flag is not set.

olerem, 2025-02-19:
Sounds plausible. Would you like to fix it? I guess, the logic in j1939_xtp_rx_rts() should be extended and session struct should be extended to store the marker for local transmission.

marckleinebudde, 2025-02-19:
We already have `j1939_tp_im_transmitter()` to check this?

olerem, 2025-02-19:
If i remember it correctly, it will be not enough.