linux-can/can-utils #602: Candump timing issue with MCP2518FD on Raspberry Pi 5

filogold, 2025-09-25:
Hi,  
I’m running into a timing problem while using MCP2518FD CAN FD controllers on a Raspberry Pi 5.

**My setup:**  
- Raspberry Pi 5 (16 GB RAM, CPU fixed at 2.4 GHz, load <20%)  
- Kernel `6.6.42-v8-16k+`  
- Two Waveshare 2CH CAN FD HATs, connected on SPI0 and SPI1  
- Overlays configured like this:
dtparam=spi=on
dtoverlay=spi1-3cs
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi0-1,interrupt=13
dtoverlay=mcp251xfd,spi1-0,interrupt=24
dtoverlay=mcp251xfd,spi1-1,interrupt=23

What I’m doing is just logging CAN traffic (the Pi isn’t transmitting). I have a setup where one ECU sends a message on CAN1, and another ECU acts as a gateway and forwards it on CAN2. Using a Vector CANcase I can see the gateway delay is always <1ms and consistent.  

On the Raspberry Pi instead:  
- With **hardware timestamps (-H in candump)** I initially see the correct order, but after some time the two channels drift apart and are no longer aligned.  
- With **software timestamps**, the relative delay between the original and the forwarded message is inconsistent (sometimes the forwarded one shows up first).  

So basically the timing correlation between the two CAN interfaces is unreliable on the Pi.

---

**Extra details I checked:**  
- Driver in use: `mcp251xfd`  
- `ethtool -g can0`:
```
Coalesce parameters for can0:
Adaptive RX: n/a  TX: n/a
stats-block-usecs: n/a
sample-interval: n/a
pkt-rate-low: n/a
pkt-rate-high: n/a

rx-usecs: n/a
rx-frames: n/a
rx-usecs-irq: 0
rx-frames-irq: 1

tx-usecs: n/a
tx-frames: n/a
tx-usecs-irq: 0
tx-frames-irq: 1

rx-usecs-low: n/a
rx-frame-low: n/a
tx-usecs-low: n/a
tx-frame-low: n/a

rx-usecs-high: n/a
rx-frame-high: n/a
tx-usecs-high: n/a
tx-frame-high: n/a

CQE mode RX: n/a  TX: n/a
```
- `ethtool -c can0`:
```
Ring parameters for can0:
Pre-set maximums:
RX:             96
RX Mini:        n/a
RX Jumbo:       n/a
TX:             16
Current hardware settings:
RX:             80
RX Mini:        n/a
RX Jumbo:       n/a
TX:             8
RX Buf Len:             n/a
CQE Size:               n/a
TX Push:        off
TCP data split: n/a
```
- Tried moving interrupts to different CPUs with `/proc/irq/*/smp_affinity`, but every time I move one IRQ, all CAN interrupts move together — I can’t split them per channel.  
- The CAN networks are quite busy but still within a reasonable busload, and the Pi CPU has plenty of headroom (<20% usage).  
- Tried to increase the priority of the process without any change.

---

### Question
- Is this just a limitation of how the driver timestamps frames (when they’re read over SPI instead of when the IRQ fires)?  
- Would reducing the RX ring size help improve determinism, or is interrupt coalescing the main factor?  
- Is it normal that the hardware timestamping drift between two MCP2518FD controllers? 
- Is there a known way to distribute the interrupts across different cores on Raspberry Pi 5?

Any suggestions on how to get stable timing correlation between two CAN interfaces on the Pi would be very helpful. Thanks!

marckleinebudde, 2025-09-25:
>     * With **hardware timestamps (-H in candump)** I initially see the correct order, but after some time the two channels drift apart and are no longer aligned.

I think each mcp2518fd chip is clocked by a separate oscillator, typically 40 MHz. The hardware timestamp is done by the mcp2518fd itself, based on the external oscillator. Over time the 2 clocks drift apart. On the Linux side the driver only converts the timestamp of the mcp2518fd into nanoseconds resolution.

>     * With **software timestamps**, the relative delay between the original and the forwarded message is inconsistent (sometimes the forwarded one shows up first).

Is this between 2 mcp2518fd on the same SPI bus? As the name suggests in this mode the timestamp is taken by the software the networking stack, not even the driver. So there's a lot of code running and scheduling done until the timestamp is finally done.

> So basically the timing correlation between the two CAN interfaces is unreliable on the Pi.

You have some options:
- HW modification: use the same oscillator for all mcp2518fd that need consistent HW timestamps. Talk to your local HW engineer if this is possible (disclaimer: I'm not a HW person)
- Daisy-Chain the mcp2518fd. Connect the 1st mcp2518fd to oscillator, connect the OSC out pin of the 1st mcp2518fd to the clock input of the 2nd one and so on. Problem: You have to hack the driver for this. I have a proof-of-concept patch series that adds a proper clock provider driver to the mcp2518fd driver. But due to limitations in the Linux Clock framework this series is not mainline.
- Add /dev/ptp support to the mcp2518fd driver and synchronize the clocks in software. I have a proof-of-concept patch series, too. But due to HW errata, reading the free running counter sometimes gives broken values, so this is not mainline either. It would requite some time to figure out good workarounds for the hardware and mainline this. This series exposes the mcp2518fd's clock directly to Linux and you can use phc2sys https://linux.die.net/man/8/phc2sys from https://github.com/richardcochran/linuxptp to synchronize the mcp2518fd clock in software against the Linux system time, or against each other.

>     * Tried moving interrupts to different CPUs with `/proc/irq/*/smp_affinity`, but every time I move one IRQ, all CAN interrupts move together — I can’t split them per channel.

Which IRQ did you move? There's the IRQ of the SPI host controller. And there's one interrupt per mcp2518fd. By the driver design these are completely independent, maybe that's a HW limitation of the RPI5.

>     * The CAN networks are quite busy but still within a reasonable busload, and the Pi CPU has plenty of headroom (<20% usage).

\o/ I spent a lot of time optimizing the mcp2518fd driver.

>     * Tried to increase the priority of the process without any change.

What does this process do? The forwarding? For "simple" forwarding from one CAN device to another, look at cangw https://manpages.debian.org/testing/can-utils/cangw.1.en.html This allows to do forwarding inside the kernel.

> ### Question
> 
>     * Is this just a limitation of how the driver timestamps frames (when they’re read over SPI instead of when the IRQ fires)?

The HW timestamp is done by the mcp2518fd chip itself. Check the datasheet which bit of the CAN frame it uses for the timestamp.

>     * Would reducing the RX ring size help improve determinism, or is interrupt coalescing the main factor?

The max FIFO size in the mcp2518fd is 32, so with the configured RX ring size of 80, the driver will use 3 independent FIFOs of size 32, 32 and 16. In burst situations, FIFO 1 might get full and the next one will be used. The driver has to bring the RX'ed CAN frames into correct order again, they get sorted by timestamp. But it's still possible that the correct order of RX messages cannot be restored. But I think you don't have this problem.

Interrupt coalescing is not activated in you output above. Coalescing will increase the performance by reducing the number of interrupts at the cost of higher latency and less accurate software timestamps.

>     * Is it normal that the hardware timestamping drift between two MCP2518FD controllers?

Yes, see above.

>     * Is there a known way to distribute the interrupts across different cores on Raspberry Pi 5?

Should be. Maybe HW limitation?

> Any suggestions on how to get stable timing correlation between two CAN interfaces on the Pi would be very helpful. Thanks!

See above.

filogold, 2025-09-26:
> Is this between 2 mcp2518fd on the same SPI bus?

In my specific case, no — I think the two mcp2518fd considered are connected to a different SPI bus.

> HW modification: use the same oscillator [...] Daisy-Chain [...] Add /dev/ptp support [...]

These hardware modifications are very interesting, but unfortunately, due to time constraints, project requirements, and also my limited hardware experience, they cannot be considered at the moment.

> Tried moving interrupts to different CPUs with `/proc/irq/*/smp_affinity`, but every time I move one IRQ, all CAN interrupts move together — I can’t split them per channel.

I tried moving the 4 IRQs, but what I noticed is that using the `smp_affinity` command I cannot split them correctly:
```
filogold@node02:~ $ cat /proc/interrupts | grep spi
119: 347919128 0 0 0 rp1_irq_chip 19 Level 1f00050000.spi
120: 769479085 0 0 0 rp1_irq_chip 20 Level 1f00054000.spi
165: 0 0 0 0 GICv2 150 Level 107d004000.spi
181: 1487 0 0 22959946 pinctrl-rp1 13 Level spi0.1
186: 394 0 0 9411911 pinctrl-rp1 25 Level spi0.0
187: 1021 0 0 26290580 pinctrl-rp1 23 Level spi1.1
188: 0 0 0 28185563 pinctrl-rp1 24 Level spi1.0
```

If I run:
```
echo 1 | sudo tee /proc/irq/181/smp_affinity
echo 2 | sudo tee /proc/irq/186/smp_affinity
echo 4 | sudo tee /proc/irq/187/smp_affinity
echo 8 | sudo tee /proc/irq/188/smp_affinity
```

the result is that all IRQs always end up on the last selected core, instead of being distributed.

> Tried to increase the priority of the process without any change. What does this process do? The forwarding?

Just to clarify: the Raspberry Pi is **not** performing the forwarding. The forwarding is done by an ECU_GW, connected to CAN1 and CAN2. ECU_GW reads the message from ECU_1 on CAN1, then retransmits it on CAN2.  
The Raspberry Pi is connected to both channels, but around 20–40% of the time I observe the forwarded message on CAN2 before the original on CAN1 (hundreds of microseconds before, sometimes 1/2 milliseconds). This is also due to the forwarding speed (the GW normally performs it in 200us). When comparing against a VN1640, I also notice a certain oscillation in the delay periods on the Pi, while the VN1640 logs show a fixed, stable delay.

---

My question is:  
Do you think that by tuning the RX ring size I could reduce or eliminate this reading order issue, or is the only reliable solution to work at the hardware level?  Do you know any other test I can do?

Thanks again for your support and for all the work you put into developing these drivers!